mainsail-crew/mainsail #2146: Add option to move the macro group Close button before the macro list

HitLuca, 2025-02-13:
### Requested feature:

The feature request relates to the Mainsail settings panel, more specifically the macros group editing. This is to avoid scrolling the huge list of available macros just to go back a step, which may happen a lot if doing some work with macros and groups. An example of what I'm referring to is below

[image]

If users don't mind the existing button placement adding a second one on the top would solve the issue, or a "To bottom ^" to go at the end of the list (less elegant in my opinion)

closed 2025-02-13 by meteyou

rackrick, 2025-02-13:
tbh I wasnt aware there is already an FR open for your request. Otherwise I had told you that on discord.

meteyou, 2025-02-13:
np and thx for the good FR description. if you have time, pls post it in the other FR.

HitLuca, 2025-02-13:
@meteyou don't understand what you mean?

meteyou, 2025-02-13:
pls post your description with screenshot in the FR #2082. FR with more content/traffic/emojis in the first post are prioritized higher.