mainsail-crew/mainsail #2148: Residual parameter when switching between two printers

Samuel-0-0, 2025-02-15:
### Mainsail Version:

edge

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When I switch from Voron 2.4 to Voron 0.2, some parameters from Voron 2.4 appear on Voron 0.2, which might be a bug.

[image]

[image]

[image]

### What did you expect to happen instead?

，

### How to reproduce this bug?

，

meteyou, 2025-02-15:
thx for your report! is this a moonraker sensor?

Samuel-0-0, 2025-02-15:
> thx for your report! is this a moonraker sensor?

yes

meteyou, 2025-02-15:
pls upload your moonraker.log/conf to reproduce the issue

Samuel-0-0, 2025-02-16:
> pls upload your moonraker.log/conf to reproduce the issue

[moonraker (2).log](https://github.com/user-attachments/files/18814923/moonraker.2.log)