mainsail-crew/mainsail #2150: [FR] Creating a way to customize the order of "Temperatures" panel in a similar way to the dashboard

AERVAQR3GZ, 2025-02-16:
### Requested feature:

Simple idea: A menu like that of the Dashboard, that can be used to reorder, select and deselect certain temperatures.  

This would be nice because it just keeps everything so ordered and in place, and is just a nice QoL touch.

[image]

[image]

### Solves the following problem:

It fixes and removes redundant temperatures and irregular orders, such as can be seen in the above attached photo, where there are 3 sensors telling the chamber temp, and the heater bed rear sensor (used to equilibrate the temperature so the whole bed is adequately warmed, not just the center) is neither above nor below the Heater Bed Center Temperature Sensor)), can be removed or reordered, cleaning up the temperature section of the dashboard. 

### Additional information:

Feel free to let me know if you need anymore information.

meteyou, 2025-02-16:
Just use a _ as first char of the name to hide objects.

AERVAQR3GZ, 2025-02-16:
> to hide objects.
Why thank you! But at the same time, it doesn't solve the reordering of the temperatures, so its not a full fix.

meteyou, 2025-02-16:
this was just a hint to "fix" one of your issues. i dont close this FR or something else. i also dont think i will implement hide, because all klipper objects can be hide with the _ and this is quite normale. thats why we dont "hide" something in the gui.

the emoji with "thumbs down" is a little bit rude. i can also just close your FR and dont implement it, if you think this is a better way?

meteyou, 2025-02-16:
and just another question. why do you use multiple times the same sensor in your printer config? (that should be the first reason, why you have multiple objects with the same value here)

AERVAQR3GZ, 2025-02-16:
> why do you use multiple times the same sensor in your printer config? 

Honestly, I don't know. The main person who has been working on the printer is my Father, and I am not that accustomed to the printer's code. I've had a look in the config, and from what I can tell, it is required to link the exhaust fans to the bed fans? Which turn on whenever the chamber heater turns on, but that might not be it. 

I tried to underscore what I thought was the variable that showed up in the dashboard, but it threw an error. I tried it with the chamber heater to see if it would work, but it said: "Unknown heater 'chamber_heater'". I think it might mean that I would have to change all of the variables with the name 'chamber_heater' to '_chamber_heater', but he has modded and coded the printer so much I am not even sure what the name of the sensors to hide are, let alone how to replace them all. 

Also, to address your question (which I now realize I did awfully little of), I think they are all separate sensors. All of the temperatures are off by a decimal of a degree at almost all times, so I think they are all unique sensors, its just that its got extra sensors from mods. 

If there are any ways that you know of to do this, I am all ears.

Many thanks

meteyou, 2025-02-19:
> I tried to underscore what I thought was the variable that showed up in the dashboard, but it threw an error. I tried it with the chamber heater to see if it would work, but it said: "Unknown heater [...]

Yes, thats right. if this name is also used in some macros or in other config sections, you have to rename it everywhere.

> Also, to address your question (which I now realize I did awfully little of), I think they are all separate sensors. All of the temperatures are off by a decimal of a degree at almost all times, so I think they are all unique sensors, its just that its got extra sensors from mods.

ah ok. i just saw 2 sensors are identical in your (exhaust & chamber), thats why i ask.

closed 2025-03-13 by github-actions[bot]