mainsail-crew/mainsail #2154: Dark mode fix

dbarwikowski, 2025-02-24:
### Mainsail Version:

Latest

### Browser:

Other (Please tell us below)

### Device:

Mobile device (Phone, Tablet or similar)

### Operating System:

Android

### What happened?

Brave browser
Dark mode Menu is hard to read

### What did you expect to happen instead?

Good dark mode menu

### How to reproduce this bug?

Open mainsail in browser with dark mode

### Additional information:

[image]

meteyou, 2025-02-24:
this is no default background image. it looks like you have a theme installed with a different sidebar-background image. If you need help to "remove" it, pls use your discord server or the discussion sections for help.

closed 2025-02-24 by meteyou

dbarwikowski, 2025-02-25:
I made installation od mainsail os on rpi zero w. I didn't change image. From what I see it is the same image but inverted colors.
[image]

meteyou, 2025-02-25:
no, the image is a little bit different. here is the original "light theme image": https://github.com/mainsail-crew/mainsail/blob/develop/public/img/sidebar-background-light.svg

your image has lines between the areas. the original one doesnt have these lines.

dbarwikowski, 2025-02-25:
There is something going on with this browser... This is the the light bg you sent

[image]

dbarwikowski, 2025-02-26:
# Solved

I had to go to the settings, disable experimental night mode and change theme to light.
Restart browser and change to dark theme. Now it looks correct :-)