mainsail-crew/mainsail #2164: Nevermore / Air Filter time Tracker Tile

Christanoid, 2025-03-07:
### Requested feature:

I have a Nevermore installed and I always change the active carbon filter material by feeling when "It's been a while".
It would be nice to have a small widget in mainsail that just tracks active print time and reports how long I have been printing with this filter inside aswell as a reset button for when I changed the filter again. This way I'd maybe not waste as much filter material switching frequently because I don't factually know if it's required.

Christanoid, 2025-03-07:
Oh neat! Never looked into the history. Good to know that this already exists. Still, I feel like this is a bit hidden / out of sight. Would an additional entry in the Miscellaneous section of the Dashboard make sense? Maybe only for repeating maintenances? Just so it's more prominent to the user that it might be coming up soon.

rackrick, 2025-03-07:
All the informationen needed for the maintenance feature depend on the print history.
Thats the reason why its placed there. The misc section is for a differnt purpose like showing light or fan information.

After creating your tasks you wont access them frequently, you just wait for the notification. So making it more prominent doesnt add any more value then just finding it.

Well I guess we should at least add it to the feature list in our docs.