mainsail-crew/mainsail #2165: Syntax highlighting issues.

Contomo, 2025-03-11:
### Mainsail Version:

v1.2.1-1-gff3869a6

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

[image]

[image]

### What did you expect to happen instead?

expected the syntax to be properly highlighted.

### How to reproduce this bug?

not sure. maybe checking if the toolchanger config has bad syntax highlighting everywhere.

### Additional information:

Two snippets of the issue, for some reason the syntax highlighting of this has been broken for almost a year. im not sure if its an issue with me or mainsail tbh.
im accessing mainsail through the orca inbuilt browser (2.3.0 dev) but the problem has been present with any orca version.
it is also present with chrome and opera.

meteyou, 2025-03-11:
there are spaces in front of the attributes. so this file is just wrong...

closed 2025-03-11 by meteyou

Contomo, 2025-03-12:
[image]
im sorry but how is this not a bug and my fault.
i mean i want it to be my fault. because that way the fix would be easy.
so if you know what is causing this. please let me know

Contomo, 2025-03-19:
cool. thanks for the great help.

meteyou, 2025-03-19:
@Contomo what do you mean?