mainsail-crew/mainsail #2187: Enable rotation of camera feed

jarruda, 2025-04-13:
### Is your feature request related to a problem? Please describe

Right now webcams are limited to hflip and vflip, and don't support 90 or 270 degree rotations if the backend doesn't support it (e.g. camera-streamer and most rpi cameras.)  This limits users to mounting cameras in landscape orientations only.

### Describe the solution you'd like

The browser is capable of performing a rotation via css, however.  You can verify this by adding `transform: rotation(270deg)` to the style of the `video` tag when using the WebRTC service. The request is to expose an option for this in crowsnest to allow users to take advantage of this.

meteyou, 2025-04-13:
to be honest. just use `transform: rotation(270deg)` is not really a solution. it would look like this:
[image]

browsers dont resize/fit an image after rotation. in mjpegstreamer-adaptive and mjpegstreamer mode, i already implemented it to rotate it to 90deg/270deg, but i have to doublecheck if it has any downside to use this for webrtc. so if you turn it 90 or 270deg, i have to resize (downsize) it to fit again here like this:
[image]

so its also not a perfect solution, but the best solution to use the same space and refit it.

mryel00, 2025-04-14:
We are aware of that rotation issue, but I didn't know about this rotation method 🤔