mainsail-crew/mainsail #2189: Information tooltip shows version but not software name

neuschaefer, 2025-04-15:
### Mainsail Version:

v2.13.2

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

- I hovered over the question mark icon in the corner of the Mainsail web interface
- This brought up a small tooltip:

[image]

This tooltip is rather difficult to understand because it only shows version numbers without context

### What did you expect to happen instead?

Each version should perhaps be accompanied by the corresponding project. "Mainsail v2.13.2" is much more useful for unfamiliar users than just "v2.13.2". I don't even know what the "🌙" icon symbolizes.

### How to reproduce this bug?

see above

neuschaefer, 2025-04-15:
Relatedly, I think this tooltip (or some other software overview page) is generally useful, because it allows an unfamiliar user to find out what this web interface is, but it's confusing if it lacks the software name.

rackrick, 2025-04-15:
The icons in front of the version number is the official logo of the software.
This is only a quick view for the three main components. Mainsail, Moonraker and Klipper.
For detailed information check the update manager on the machine site. It shows the name of the software, current version, available version, etc.
In addition other components which are not part of this main software trinity of the tooltip are also shown.

Bloating up the tooltip is not really necessary in my opinion

neuschaefer, 2025-04-15:
I think this is potentially confusing for one scenario: A new and unfamiliar user uses an installation of Mainsail for the first time and tries to find out what the web interface is. Such a user cannot look up the icons because they don't easily translate into words that could be entered into a search engine.

The Machine page also doesn't explain the icons, because it doesn't show them, and the version overview is hidden under the (seemingly) unrelated title "System loads".

Fortunately, I found the word Mainsail in the title, so my confusion wasn't forever. I just think this is a suboptimal experience for new users. 

Of course any change has to be balanced against the quality of experience for seasoned users, and I'll leave that up to you.

rackrick, 2025-04-15:
> The Machine page also doesn't explain the icons, because it doesn't show them, and the version overview is hidden under the (seemingly) unrelated title "System loads".

That seemingly unrelated title is just a different panel which happens to be positioned above the update manager. Which is its own panel.

But... I get your point we're not really showing "this is mainsail" somewhere. For new users who just see a finished setup it might really hard to find out which interface this actually is. 
Adding the names to the version tooltip is just as hidden as the browser title. I think we should come up with a more prominent place to show this.

Maybe something like "Welcome to mainsail" banner on the init screen or similar. Thanks for you feedback we give a thought and come up with something

neuschaefer, 2025-04-15:
> I think we should come up with a more prominent place to show this.
> 
> Maybe something like "Welcome to mainsail" banner on the init screen or similar. Thanks for you feedback we give a thought and come up with something

Perhaps on the start page / printer selection (which might be what you mean by init screen):

[image]

neuschaefer, 2025-04-15:
> That seemingly unrelated title is just a different panel which happens to be positioned above the update manager. Which is its own panel.

Ah sorry, I got confused. I saw some version numbers in _System Loads_ (for MCU firmware and some related component on the host) and thought this was the place.

meteyou, 2025-04-16:
you are right. we doesnt show the name here, because it would make the panel too big. normally the update manager will be used to display version numbers, because it contains all components. This tooltips is just a "fallback" for instances without update manager. I think it is also older than the update manager.

To be honest... If you setup your system and you choose Mainsail to install it and flash your original printer with Klipper, you should know what you use...

The Select Printer dialog is no default dialog. This only exists on multi instances (that is rather rare) and per default, if you dont set printer names, you should see instead of the Printer name "Mainsail" (the fallback name) on the top right where the Mainsail logo is.

neuschaefer, 2025-04-16:
> To be honest... If you setup your system and you choose Mainsail to install it and flash your original printer with Klipper, you should know what you use...

_If_..., then yes. But there are also shared installations in which not every potential user is the admin who installed Mainsail. It could be argued, perhaps, that it's the local admin's responsibility to document the software that was used in that case.