mainsail-crew/mainsail #2219: Using Scarf Seams with Orca Slicer causes a rendering issue with the G-Code Viewer

jaron780, 2025-05-19:
### Mainsail Version:

v2.14.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When using Orca Slicer with the Scarf Seams feature enabled. The outputted G-Code Causes a rendering issue with Mainsail's G-Code Viewer. Specifically it causes the layers to appear super thin instead of the proper layer height.

This is with Scarf seams Enabled:
[image]

This is with Scarf seams Disabled:
[image]

This issue still happens even when using the Start gcode of:
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]

And using the Layer Change gcode of:
SET_PRINT_STATS_INFO CURRENT_LAYER={layer_num + 1}

I assume this is caused because the scarf seams produce gcode that slowly/incrementally goes up along the z axis while moving and isn't a clear z transition making the G-Code viewer treat the last z movement as the previous layer making the next layer seem extremely thin.

### What did you expect to happen instead?

I expected the G-Code viewer to show the same rendering as with scarf seams disabled.

### How to reproduce this bug?

Use Orca slicer and Enable Scarf Seams and upload it through mainsail and view it with the 3D G-Code viewer and the layers will appear extremely thin.

meteyou, 2025-05-19:
Thx for your report, but we are using an external lib for this gcode viewer and it looks like this feature is not included in this lib. i will send this bug to the maintainer of this lib.

jaron780, 2025-05-19:
Here is gcode with Scarf seams enabled and disabled. hope that helps!

Enabled:
[Scarf Seams Enabled.txt](https://github.com/user-attachments/files/20298457/Scarf.Seams.Enabled.txt)

Disabled:
[Scarf Seams Disabled.txt](https://github.com/user-attachments/files/20298459/Scarf.Seams.Disabled.txt)

> [@jaron780](https://github.com/jaron780) I am the dev who wrote the viewer. I am going to take a look at this. Can you provide a sample g code file that I can look at.
> 
> Thank you!

PR #2227 (2025-06-14): fix(gcodeviewer): fix scarf seams in Gcodeviewer
## Description

This PR fix Scarf Seams in the Gcodeviewer by bump `@sindarius/gcodeviewer` package to 3.7.17. Thx @Sindarius for fixing this bug!

## Related Tickets & Documents

fixes #2219 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2025-06-14 by meteyou