mainsail-crew/mainsail #2239: [FR] Display inductive-probe threshold (z_offset) somewhere on dashboard

eric-s-raymond, 2025-07-18:
### Requested feature:

Please add a display of the current saved z_offset on the Dashboard somewhere.

At present, there are a couple of ways of checking it that are all a bit circuitous and a hassle.

### Solves the following problem:

Troubleshooting elephant foot and other z-offset problems is more work than it should be.

meteyou, 2025-07-18:
@eric-s-raymond Thanks for reaching out with your feature request.

I'd appreciate it if you could elaborate a bit more on *why* you need to see the current z-offset (probe) on the dashboard/interface. Especially with an inductive probe, this value represents the distance between an "invisible point" in front of the nozzle and the nozzle itself. Simply seeing this number, in my opinion, doesn't inherently provide a positive value to the user, as it's purely a "fictional" reference point.

Regarding your point about troubleshooting elephant's foot, while Z-offset can certainly contribute, this issue can also stem from several other factors beyond just the precise Z position of the first layer.

Could you provide more context on how displaying this specific value would aid in your troubleshooting or daily use?

closed 2025-07-20 by meteyou

eric-s-raymond, 2025-07-20:
I protest this.  Whether I misunderstand z_offset or not, refusing to display the saved value is a deficiency in Mainsail because iit hides an important piece of Klipper state that make tuning the printer's behavior unreasonablty difficult, and forces awkward workarounds.