mainsail-crew/mainsail #2265: Add an Dashboard Panel for ACE Pro

DerDoubleD, 2025-09-20:
### Requested feature:

View ACE Pro temperature
ACE Pro filament management
ACE Pro configuration
Start drying process

[image]

### Solves the following problem:

Very Nice to have!
You no longer need the Next-Slicer

### Additional information:

```
**Get information about the box**
curl -X POST -H "Content-Type: application/json" -d '{}' http://192.168.xxx.xxx:7125/printer/filament_hub/info

**Get values about the box**
curl -X POST -H "Content-Type: application/json" -d '{}' http://192.168.xxx.xxx:7125/printer/objects/query?filament_hub

**Get information on consumables**
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "index": 0}' http://192.168.xxx.xxx:7125/printer/filament_hub/filament_info
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "index": 1}' http://192.168.xxx.xxx:7125/printer/filament_hub/filament_info
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "index": 2}' http://192.168.xxx.xxx:7125/printer/filament_hub/filament_info
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "index": 3}' http://192.168.xxx.xxx:7125/printer/filament_hub/filament_info

**Turn on drying-function**
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "temp": 60, "fan_speed": 50.0, "duration": 2}' http://192.168.xxx.xxx:7125/printer/filament_hub/start_drying

**Stop drying function**
curl -X POST -H "Content-Type: application/json" -d '{"id": 0}' http://192.168.xxx.xxx:7125/printer/filament_hub/stop_drying

**Get Feed box configuration**
curl -X POST -H "Content-Type: application/json" -d '{}' http://192.168.xxx.xxx:7125/printer/filament_hub/get_config

**Set Feed box configuration**
curl -X POST -H "Content-Type: application/json" -d '{"auto_refill": 0, "flush_multiplier": 0.5, "flush_multiplier_editable": 0.9, "flush_volume_max": 801, "flush_volume_min": 105, "runout_detect": 1}' http://192.168.xxx.xxx:7125/printer/filament_hub/set_config

```

meteyou, 2025-09-21:
hey! thx for your request, but something looks wrong to the urls. Normally Moonraker (the port 7125 is Moonraker) only have /printer/query and /printer/list urls. Do you have a link from the moonraker plugin you need for an ACE Pro?

swilsonnc, 2026-03-23:
In the meantime, if anyone is looking for a mainsail panel for the Ace Pro you can check out my driver which includes mainsail and fluidd panels at [https://github.com/swilsonnc/ACEPROK1Max](https://github.com/swilsonnc/ACEPROK1Max)